Hypfer/esp8266-vindriktning-particle-sensor #1: Temperature and Humidity

GSzabados, 2021-07-17:
Hello,

First of all, awesome project. I didn’t even know that IKEA has an Air Quality sensor. Some might would think that they would fit inside their Air purifier first a sensor like this...

Anyhow, have you thought of adding a Temperature and Humidity sensor into the device to give a full range of environmental details? Like a DTH22 or AM2302. Is there enough space to fir that as well?

Best Regards,

Gábor

Hypfer, 2021-07-19:
As I personally already have a functioning bunch of zigbee temperature sensors basically everywhere, I did not try that, however it should be an easy modification software-wise.

As for hardware, it might get a bit too crammed in there.
Note that adding too much stuff in there might interfere with the airflow and therefore mess up the readings

GSzabados, 2021-07-19:
> Note that adding too much stuff in there might interfere with the airflow and therefore mess up the readings

Thanks for the reply. I've just had some time this morning and checked the images of the device from Twitter. I understand now what you mean regarding the airflow. Anyhow I will give it a try with a DTH22 HAT. I ordered this morning before checking the pictures. I will keep you updated how it worked out. (I might will need some help for the sketch as I don't have too much experience with it.)

Hypfer, 2021-07-19:
If you want to extend this codebase, you probably would want to do the following

Extend that state struct to also contain the data from the DHT22
https://github.com/Hypfer/esp8266-vindriktning-particle-sensor/blob/320cdc01117ec18da23eded5a7450d70119d8cdd/src/types.h#L1-L5

Extend the autoconfig section to also publish temperature and humidity sensors

https://github.com/Hypfer/esp8266-vindriktning-particle-sensor/blob/320cdc01117ec18da23eded5a7450d70119d8cdd/src/esp8266-vindriktning-particle-sensor.ino#L261-L274

Extend the MQTT publish function to add the other measurements to the published JSON

https://github.com/Hypfer/esp8266-vindriktning-particle-sensor/blob/320cdc01117ec18da23eded5a7450d70119d8cdd/src/esp8266-vindriktning-particle-sensor.ino#L217-L219

And since the DHT22 requires polling afaik, extend the loop function to poll it on each publish

https://github.com/Hypfer/esp8266-vindriktning-particle-sensor/blob/320cdc01117ec18da23eded5a7450d70119d8cdd/src/esp8266-vindriktning-particle-sensor.ino#L141-L148

That should be about it

closed 2021-07-21 by Hypfer

matrob, 2021-08-16:
The additional function of the DHT22 would be absolutely great. Is anyone already working on this?

GSzabados, 2021-08-16:
> The additional function of the DHT22 would be absolutely great. Is anyone already working on this?

I went down the ESPHome way at the end as @Habbie made a ESPHome component which is/will be part of the latest release of ESPHome.

matrob, 2021-08-16:
Thank you for pointing this out. I will take a closer look at his PR https://github.com/esphome/esphome/pull/2038.

unreal4u, 2021-08-16:
Hi all!

Based on above indications of @Hypfer (hey, thanks for your work!), I created a quick-'n-dirty© commit that implements a DHT11 sensor. Please adjust accordingly if you are using the DHT22 variant :)

I connected it to D4 of the Wemos Mini, or GPIO02, adjust accordingly.

Because it is so quick and dirty, I won't create a PR: ideally I would want to define one variable in order to include all the additional information + libraries and such, but this is the first time I have played with an ESP8266 board, so I def. will have done some very wrong stuff!

The commit can be found at my fork:
https://github.com/unreal4u/esp8266-vindriktning-particle-sensor/tree/support-for-dht-sensor

To compile, you'll need the Adafruit Unified Sensor (v1.1.4) library installed. I also installed the DHT Sensor Library (v1.4.2) but I'm not quite sure if this is needed or not.

grmcdorman, 2021-12-30:
I have an alternative firmware implementation, https://github.com/grmcdorman/esp8266_vindriktning; this uses an SHT31-D (an I2C device), but it shouldn't be hard to add the DHT variants to the code base from the way I've structured it. I have it working nicely with the SHT31-D.

The Vindriktning polling in that code has been gently revised; I believe it is more robust (for example, that 25 ms wait "for unknown reasons" isn't needed any more).