Hypfer/esp8266-vindriktning-particle-sensor #2: Crash after first reading is received

joelhaasnoot, 2021-07-17:
Thanks for this excellent work! 

I tried to replicate but after receiving the first reading, the system crashes. A side-effect of this is that the Wifi Config is lost and you have to reconfigure it, making debugging a bit of a pain :)

The serial log looks like this:
```
IP: 192.168.1.239
-- Current GPIO Configuration --
PIN_UART_RX: 4
Receiving:................................................................Receiving:....................Done.
Received PM 2.5 reading: 6

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (0):
epc1=0x40201673 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffdc0 end: 3fffffc0 offset: 0190
3fffff50:  3ffef6a8 3ffef9a4 00000004 40201673  
3fffff60:  40217c3d 0000000f 00000000 00000000  
3fffff70:  3ffef6a8 3ffef720 3ffef1c0 402037cc  
3fffff80:  00000000 00000000 00000001 3ffefb34  
3fffff90:  3fffdad0 00000000 3ffefb20 40203800  
3fffffa0:  3fffdad0 00000000 3ffefb20 40217640  
3fffffb0:  feefeffe feefeffe 3ffe862c 4010156d  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------```

I've decoded the exception to be
```
Exception 0: Illegal instruction
PC: 0x40201673
EXCVADDR: 0x00000000

Decoding stack results
0x40217c3d: __delay(unsigned long) at /Users/user/Library/Arduino15/packages/esp8266/hardware/esp8266/3.0.1/cores/esp8266/core_esp8266_wiring.cpp line 55
0x402037cc: handleUart() at /Users/user/Downloads/esp8266-vindriktning-particle-sensor-master/src/esp8266-vindriktning-particle-sensor/serialCommunication.ino line 81
0x40203800: loop() at /Users/user/Downloads/esp8266-vindriktning-particle-sensor-master/src/esp8266-vindriktning-particle-sensor/esp8266-vindriktning-particle-sensor.ino line 142
0x40217640: loop_wrapper() at /Users/user/Library/Arduino15/packages/esp8266/hardware/esp8266/3.0.1/cores/esp8266/core_esp8266_main.cpp line 201
```

(using a D1 Mini Pro)
Any ideas what's going wrong?

joelhaasnoot, 2021-07-17:
Ah, took a little longer and figure it out. Once I had downgraded to ESP8266 Board/Core version 2.7.4 it works fine. 

It does however still seem to constantly be losing its wifi settings.

Hypfer, 2021-07-19:
Huh. Interesting. Unfortunately, I can't say much about both of your issues since I honestly lack the experience.

Which version of ESP8266 Board/Core were you using before? Maybe there are breaking changes mentioned in the release notes between 2.7.4 and that version.

In any case lets hope that someone with the suitable skills reads this issue and shines some light on the problem

joelhaasnoot, 2021-07-21:
I used 3.0.1 before. 
I think the issue may be timing related. It seems to be quite sensitive timing wise, see my other message.

joelhaasnoot, 2021-07-21:
> It does however still seem to constantly be losing its wifi settings.
This was an issue on my side

enwi, 2021-07-22:
I think both issues (#2 and #8) are related to the `parseState()` function, which was declared to return a bool, but did not return anywhere, which will result in undefined behaviour. This has been fixed with #9 and I think it is already working for @joelhaasnoot again.

https://github.com/Hypfer/esp8266-vindriktning-particle-sensor/blob/e6ff884dfb68036f855f042d9b4e05165366b09d/src/serialCommunication.ino#L1

Hypfer, 2021-07-23:
Thanks for figuring that out @enwi !

As I just copy-pasted most of this code from one project to another, the same issue did surface here:
https://github.com/Hypfer/esp8266-midea-dehumidifier/commit/fabc01297fe6c787b87e095390be8abcaafbc91c

I am somewhat surprised that the compiler didn't complain though

enwi, 2021-07-23:
@Hypfer C compilers can sometimes be strange. I mean I didn't even get a compiler warning, so I first thought the `memcpy` did something weird. But it wasn't until I tried to force the function to return by adding a return statement that the static analysis (intellisense) told me, hey you can't just return you need to return a boolean that I noticed it.

closed 2021-07-24 by Hypfer