Hypfer/esp8266-vindriktning-particle-sensor #4: Not an issue - just a question! ZigBee instead?

LaurenceGough, 2021-07-20:
Hi, thanks for the project, very clever idea, and an interesting device. 

I did purchase one of these sensors today, I didn't even realise it had a fan, quite a noisy one at that! You're right, without the fan operating such as generating some smoke behind it it doesn't detect it at all, but as soon as the fan turns on it does just fine. I do wondering if lowering the voltage on the fan gives the same accuracy, because I sure am tempted! Doing some very caveman style testing shows it's very accurate so far.

There is some interesting info on the sensor if it is this one. Interesting the range of PM it can detect, not just 2.5um particles, unless this is a custom modification made! I wonder if it can detect grass pollen at around 20um
https://pdf.directindustry.com/pdf/cubic-sensor-instrument-co-ltd/pm1006k-led-particle-sensor-module/54752-927719-_4.html

Anyway, onto my question - if I wished to do a similar mod but using ZigBee (Smartthings hub) rather than WiFi would you have any suggestions?

This devices begs for smart connectivity, I can't believe Ikea did not do this with their already established ZigBee range!

Many thanks

Hypfer, 2021-07-20:
I'm not aware of any zigbee-capable uCs that are as easy to work with as the esp8266.
I'm sure that there must be _something_ but it will either be expensive or hard to work with or maybe even both.

This is made worse by the fact that the SmartThings Hub is a proprietary thing which you cannot extend and/or debug in any way.
It wouldn't surprise me if they also have a whitelist of allowed devices.

I am not an expert on zigbee devices, however looking at both the [ZCL](https://www.nxp.com/docs/en/reference-manual/ZCLRM.pdf) as well as the [ZHA](https://www.nxp.com/docs/en/user-guide/JN-UG-3076.pdf) spec there doesn't seem to be some existing structure for air quality measurements.

If I understand it correctly, in theory it should be possible to extend ZCL with custom functionality. However as you don't control the code of your Zigbee coordinator, this is definitely impossible.

closed 2021-07-24 by Hypfer

MattWestb, 2021-08-29:
Sorry for jumping in but Starkvind that is being reeled in October is having one Zigbee module (the new EFR32MG21) and the OTA firmware is released and can being flashed on other EFR32MG21 devices.
I see 2 version they can have doing it:
1. Using the vindriktning PCB and the debug connection for "sniffing" the sensor and using the fan connection (not populated on the PCB) for steering the fan speed (if not loading one different firmware in the MCU).
2. Making one new PCB and have connecting the sensor to it and have one power supply and circuits for PWM steering of the fan (for the purifier) and also the steering knob and is using the Zigbee module MCU for doing all work.

Very likely they have implanting the Zigbee cluster for all relevant data transmitted to there host system and the RX from the sensor and perhaps sending the commands for doing the measuring if not using the vindriktnings PCB.