Hypfer/esp8266-vindriktning-particle-sensor #5: Not an issue but a suggestion: Merge into Esphome?

oopsitsaflame, 2021-07-20:
Hi, would it be possible to get this device supported into esphome? I think it would fit quite good as a supported sensor and you would get all the other stuff esphome supports as a bonus.

Hypfer, 2021-07-20:
@habbie already did that
See https://github.com/Hypfer/esp8266-vindriktning-particle-sensor/issues/3#issuecomment-882858203

Personally, I don't use ESPHome due to philosophical differences.

closed 2021-07-20 by Hypfer

Hypfer, 2021-07-20:
Just for reference, I'd like to mention that decentralization is a good thing.
We shouldn't be merging everything into a single yaml behemoth.

Nabu Casa is slowly eating the smarthome world which by definition is a bad thing even though their intentions are most likely good.
We need better solutions to eliminating boilerplate code than merging everything into a single legal entity.

oopsitsaflame, 2021-07-20:
Just for reference too ;) I'd like to mention that while your concerns are understandable., I just like not having to rely on projects where I have to search half the internet for and then to discover that the one dude/duddette who  maintained it alone has last logged in 4 years ago. (not talking about you, but in general). Open source can help with both these problems but not eleminate them completely. 
https://xkcd.com/2347/