Hypfer/esp8266-vindriktning-particle-sensor #10: Question about wiring points

wizbowes, 2021-07-21:
The photo you have of the wired up test points, you seem to have the data wire connected to the test point labelled 'REST'.  Just confirming that this is correct.  Isn't REST a reset line of some kind, and the far left test point labelled ISPDA the data test point?  Or is the board labelling incorrect.  Don't have a scope so can't really check myself...

Hypfer, 2021-07-21:
Yes that is correct.

The board looks like it might've also been used in a different configuration. I suspect that that's why the label isn't UART_TX or something like that.

closed 2021-07-21 by Hypfer

trejan, 2021-07-23:
The labels don't match normal operation functions because those test points are for in-circuit programming of the uC. It needs ISPCL, ISPDA, MRSTN (REST), VDD and VSS which are all present in that row. The MRSTN pin is configured to be UART RX.

drthanwho, 2023-01-15:
Is there any chance that by accidentally first using ISPDA instead of REST that the board was damaged? I confused myself and connected D2 to ISPDA, realised because was getting an `unknown` value, connected it to REST but still getting no values