Hypfer/esp8266-vindriktning-particle-sensor #7: Not an issue - A little more information about the fan

sermayoral, 2021-07-21:
Hi. First of all, thanks for this amazing project! I really want to do it, and during the analysis, I had some questions regarding the fan.

If I understand correctly, the firmware of the Ikea board toggle the fan every minute. Surely the firmware does a reading with the fan on. The doubt is the next one: If you replace the 5v output of the ikea board for the 3v3 output of the esp. Does the toogle function hold? Or would you have a fan that is always running?

Thanks

Hypfer, 2021-07-21:
That results in the fan running constantly as it completely bypasses the FAN outputs of the Ikea PCB

sermayoral, 2021-07-21:
Thanks @Hypfer.

What do you think about applying a step down to the FAN outputs of the Ikea PCB in order to reduce voltage to 3.3V? In that way the toggle function should work.

Hypfer, 2021-07-21:
That would require more hardware. The permanent 3.3V were just there for the taking
I wasn't able to spot any downsides of running the fan continuously

Actually I'm more worried about the 5V UART to the 3.3V ESP8266 regarding long-term reliability
Only time will tell

sermayoral, 2021-07-21:
Thanks @Hypfer for the information. I will test with the step down and I will share the results here :-)

Habbie, 2021-08-16:
> Actually I'm more worried about the 5V UART to the 3.3V ESP8266 regarding long-term reliability

I did extensive Googling on this and it appears the ESP8266 UART is 5V safe for receiving.

trlafleur, 2021-10-02:
I use a ~470ohm resistor from the IKEA to the Wemos-D1 Mini pin D7 (GPIO-13) with a surface mount Schottky diode (they have a low V-Forward drop) from D7 with its cathode connected to +3.3v, this clamp the voltage to 3.3V + V-forward drop of the diode... this makes it safer for the ESP8266 (I now see 3.7V on my unit on D7)... Using pin D7 makes it an easy fit for the diode to 3.3v pin on the Wemos-D1 circuit board. ( 3.3v - D8 - D7 ......) so the diode is set on top of D8, with just enough room to solder it to 3.3V and D7.

onshisan, 2022-01-01:
> Just as a note: I move my unit from this codebase to Tasmota

Sounds great, I was also thinking of adding more so I'm curious... any notable issues to report using Tasmota to accomplish this?

trlafleur, 2022-01-01:
useing an esp32, works just fine....

~~ _/) ~~~~ _/) ~~~~ _/) ~~~~ _/) ~~

Tom Lafleur