Hypfer/esp8266-vindriktning-particle-sensor #8: Sensor disappears in Home Assistant

joelhaasnoot, 2021-07-21:
Have been running this for a few days, but the sensor disappears quite often. 
I'm guessing it's timing related - where the MQTT loses it's connection and a last-will message is sent.

Hypfer, 2021-07-21:
Might be related to this:
https://github.com/knolleary/pubsubclient/issues/239

and this:
https://github.com/Hypfer/esp8266-vindriktning-particle-sensor/blob/e6ff884dfb68036f855f042d9b4e05165366b09d/src/esp8266-vindriktning-particle-sensor.ino#L90

Please check your broker config.

joelhaasnoot, 2021-07-21:
Sort of, but `mqttClient.loop();` should generally be able to be called every 10 seconds. Or does reading the serial connection take that long?

Hypfer, 2021-07-21:
Yes I don't think it's the firmware itself dying but rather the wifi connection.

It might reconnect faster than the older connection times out which means that the LWT will be sent after the new connection has publishedb the "available" status.