Hypfer/esp8266-vindriktning-particle-sensor #12: Potential problem with running the fan at 3.3V?

Leeev, 2021-07-24:
I think there may be an issue with running the fan at 3.3V.
At 5V the fan will push a certain CFM for certain amount of time.
At 3.3V the fan will push much less CFM for that time.
The sensor unit counts the number of particles that are pushed though it in that time.
Net result, the sensor will see many less particles (due to less volume of air passing through it) and will under report the air quality.

Hypfer, 2021-07-24:
> The sensor unit counts the number of particles that are pushed though it in that time.

Nope. It counts the particles in the air that it sees. As they are distributed evenly in every m³ of air, the measurement is not impacted. At least I couldn't see any difference in readings when put side by side.

It is possible that the sensor will react slightly slower to changing conditions though, as on change, the new amount of particles may not be distributed as evenly. That however is a temporary state that resolves itself pretty quick.

closed 2021-07-24 by Hypfer

Hypfer, 2021-07-24:
[image]
[image]

It is still rather quick to react IMO. Someone in the neighbourhood started to burn something just now

Leeev, 2021-07-27:
I see what you're saying, but I think you may be incorrect. The firmware in the unit is calibrated to a full fan speed.
At full speed the fan could move, say, 100cm3 of air through the unit, in say,10 secs. If there are 1000 particles in that 100cm3 of air the unit will register 1000 particles. 
If the fan is working at 3.3V, it will move, say, 2/3 of that 100cm3 of air through the sensor in those 10 secs and therefore only register 666 particles. 
This would suggest that the air is 'better quality' that it really is?
I'm not say it won't work, but rather the data will be low/incorrect.

Hypfer, 2021-07-27:
As discussed on hackaday I'd argue that that is not the case as there is no mention of the fan and/or its requirements in the datasheet of the sensor.

The fan is not part of the sensor but something that was bolted on by Ikea.
Therefore, I find it hard to believe that anything _inside_ the sensor depends on something purely external