Hypfer/esp8266-vindriktning-particle-sensor #15: UART levels 5V / 3.3V

bertmelis, 2021-07-30:
You write in your readme `"ESP8266 is 5V-tolerant, this should there shouldn't be any issues"`. This is not true of course. Anything aboce 3.6V can damage the pins.

Since the UART level of an PM1006 is I do expect issues with burning pins on the ESP8266.

Maybe the IKEA board has a level shifter? Did you find one?

Hypfer, 2021-07-30:
> This is not true of course. Anything aboce 3.6V can damage the pins.

https://twitter.com/ba0sh1/status/759239169071837184

https://hackaday.com/2016/07/28/ask-hackaday-is-the-esp8266-5v-tolerant/

bertmelis, 2021-07-30:
🤯

closed 2021-07-30 by bertmelis

Hypfer, 2021-07-30:
I'm actually curious if there are any long-term side-effects. As of now I haven't seen any but that doesn't mean much

bertmelis, 2021-07-30:
Went to Ikea yesterday, hacking this evening. I might throw in a voltage divider though, just to be sure. I'm short on devices atm.

bertmelis, 2021-08-05:
so I tried with a simple voltage divider (2k2, 3k3) and this happens:
it works when I power the whole module via the Wemos D1 it works flawlessly. When I power via the main USB-C port it doesn't work. That is: it does connect to MQTT but the SoftwareSerial doesn't detect anything.

My electronics skills are a bit rusty. Would you have a clue?

bertmelis, 2021-08-05:
Nevermind. Got it. I guess I ruined a path when soldering.

I used another pin and now it works without a problem.

bilogic, 2021-11-17:
I did some googling.

For folks like me who are not hardware savvy, the term 5V tolerant may be interpreted as being able to use 5V to power the chip, i.e. directly wire the USB power to ESP8266's 3.3V

But that does not seem to be the case as clarified by the CEO of Expressif here https://www.facebook.com/groups/1499045113679103/posts/1731855033731442/?comment_id=1732364133680532

- 5V I/O - ok
- 5V power - not ok

It might be better to state that clearly: `As the ESP8266 I/O pins are 5V-tolerant, this should there shouldn't be any issues...`

Thanks.

LordPinhead, 2021-11-17:
Hi, so that's a true and false thing. The chip itself is not 5V safe, you have to use a regulator of some kind. But the D1 has  this on board.
If the regulator is not damaged, you can power it safely with USB. If its damaged though, you could destroy the chip.

I ordered 10 D1 Mini from Aliexpress and they were all goners, 1 from Germany that works without a hitch.
I measured the Vin  of the ESP and it had between 1 and 3 Volts jumping or worse, 5V directly. 

So a few are really gone, a few I could maybe safe when use a buck converter. But it's true that the esp chip itself will die with more than 3.6V in.

You know its dead when you flash the sketch and it constantly reboots with strange exceptions. Measure VIN on the esp and see if the regulator of the board is toast.

bilogic, 2021-11-17:
I commented earlier because I almost connected 5V to my 3.3V rail before "common" sense kicked in and I googled. I'm sure some have blown their ESP8266 due to misinterpreting "5V tolerant".

Even if some ESP8266 can endure the 5V, most won't. All I'm trying to do here is to prevent more damage.