Hypfer/esp8266-vindriktning-particle-sensor #18: Control LED? Night mode?

fabiosoft, 2021-08-04:
Did you find same traces for LED controlling? I can see some pins labelled LED_G and LED_R...but i can't figure out what's the purpose and if i can use GPIO or need a relay... btw where is the Yellow? 🤣

Maybe could be useful in bedrooms during the night (too tiny space to add a photoresistor too?)

Hypfer, 2021-08-04:
There already is a light sensor for automatic dimming in place, however even in the "dim" state, the LEDs are still way too bright IMO.

I've mitigated that issue with my sensors by taking a black permanent marker and adding a few layers of paint to the LEDs.
They're still a bit too bright though

fabiosoft, 2021-08-04:
So next step... cut out uC for led 😇  ...and use pins and transistors/relays to control colors with a rule based on epa rules (already followed by original fw) [AIR QUALITY STANDARDS (page 4)](https://www.epa.gov/sites/default/files/2016-04/documents/2012_aqi_factsheet.pdf)

brev-dev, 2021-08-05:
I believe the image below shows the resistors corresponding to the LED pairs (greens are in parallel with two resistors; red and yellow are in series, with one each). So you should be able to tweak their values to reduce the offensive glare. I've popped-off R9 as a quick way to halve the green output. The least destructive alternative might be to shove some paper behind the frosted window.
[image]

mcodebase, 2023-04-09:
Hi,
I try the solution from SnowHead (add a resitor in parallel to R8 (22k)), 
this resistor will be reduce the brightness when it's bright to the "dark" mode.
But no chance to reduce the "dark mode" darker ...

My question, is there a Pin or an option to cut a ciruit path physically, so the LED will be completly "off".
My idea is to add a switch or relay so the Vindrikning is completly dark in the night (but still woking).

xSnowHeadx, 2023-04-11:
There is no simple solution for that. You only can switch the LED completely off by removing the connections of the red and green LED to the positive voltage and of the orange LED to ground by scratching the copper lines and using a dual poled switch to reconnect them on demand.

dudzio12, 2024-11-07:
@mcodebase have you found any solutions? :)