Hypfer/esp8266-vindriktning-particle-sensor #20: Not an issue - Video on how to convert and setup the particle sensor

enwi, 2021-08-07:
Hey @Hypfer I made a video on how to convert the Vindriktning sensor and how to set it up. I you want you can add it to the readme, as it will go through all the needed steps. There is also a German version of the video, which is also linked from the English version 😄 
https://www.youtube.com/watch?v=NwHL8XMarFY

Hypfer, 2021-08-07:
https://github.com/Hypfer/esp8266-vindriktning-particle-sensor/issues/15#issuecomment-889892467

enwi, 2021-08-07:
I would still go for the extra protection using the voltage divider, as I remember correctly I killed an ESP8266 before supplying 5V to a GPIO. So better stay on the safe side.

Q7Jensen, 2021-08-08:
When the device is configured, how can I reconfigure it a second time to select another mqtt broker?

enwi, 2021-08-08:
@Q7Jensen that's still a [todo](https://github.com/Hypfer/esp8266-vindriktning-particle-sensor#todo)