Hypfer/esp8266-vindriktning-particle-sensor #23: get no values from sensor

caveman2003, 2021-08-11:
Hi all, 

after install and successful upload to wemos i can see messages in MQTT Explorer but no values ... i just see:
```
{
  "device": {
    "identifiers": [
      "VINDRIKTNING-49ED42"
    ],
    "manufacturer": "Ikea",
    "model": "VINDRIKTNING",
    "name": "VINDRIKTNING-49ED42",
    "sw_version": "2021.08.0"
  },
  "availability_topic": "esp8266-vindriktning-particle-sensor/VINDRIKTNING-49ED42/status",
  "state_topic": "esp8266-vindriktning-particle-sensor/VINDRIKTNING-49ED42/state",
  "name": "VINDRIKTNING-49ED42 PM 2.5",
  "unit_of_measurement": "μg/m³",
  "value_template": "{{value_json.pm25}}",
  "unique_id": "VINDRIKTNING-49ED42_pm25",
  "icon": "mdi:air-filter"
}
```
i would expect to see some values in "value_template" and not  "{{value_json.pm25}}"... i would say i solder correctly on the board ... also i tested with PIN_UART_RX = 5 in SerialCom.h but the same output ... is my wemos maybe brocken ? 
any ideea?

Thanks .... 
GREAT IDEEA :-)

caveman2003, 2021-08-11:
or would i see an value here only if the sensor detect something? i tried so simulate some "dust" but LED is still green :-)

brev-dev, 2021-08-13:
Issue [#16](https://github.com/Hypfer/esp8266-vindriktning-particle-sensor/issues/16) might be relevant for you. It seems some sensors are calibrated such that for most of the time they only produce zeros. To debug what's happening in your case, I suggest you focus on seeing what happens when the color is changing on the device: when the device is turning to at least yellow, that's your independent verification that the wemos should be detecting valid readings.

Dust might be too coarse-grained. Blow some smoke over it, from a match for example. Stick it under a bowl if you want the smoke to linger.

caveman2003, 2021-08-16:
Hi brew-dev,

thanks for the hint with the match ... :-) bad that i'm not smoking ... but an small piece of paper produce also some smoke :-)

the esp8266 was broken after changing the device i get:

`{
  "pm25": 1000,
  "wifi": {
    "ssid": "xxxxx",
    "ip": "192.168.xxx.yyy",
    "rssi": -77
  }
}`

So it works perfectly!!!  THANKS!!

closed 2021-08-16 by caveman2003