Hypfer/esp8266-vindriktning-particle-sensor #24: which libraries to use? Just a question.

MHz000, 2021-08-13:
- working with VScode I'm insecure which libraries to use.
lib_deps = 
	bblanchon/ArduinoJson@^6.18.3
	tzapu/WiFiManager@^0.16.0
	knolleary/PubSubClient@^2.8

- Config.h,Types.h and SerialCom.h already in scr folder but "compilation terminated ... include files not found" 

- besides the 3.3V issues would an ESP-01 be sufficient
Thank-you for advice

LordPinhead, 2021-11-06:
When you open the Sketch in Arduino Studio and import the sketch (ino) it says the sketch has to be in a subfolder, then it creates the subfolder and moves the Ino file into it. So "src/esp8266-vindriktning-particle-sensor.ino" will become "esp8266-vindriktning-particle-sensor/esp8266-vindriktning-particle-sensor.ino" and then the files have to be copied over to the folder.

That helps with that issue.

zfsnotification, 2022-02-13:
I can't get past SerialCom.h not found  I have a copy in every folder