Hypfer/esp8266-vindriktning-particle-sensor #38: Control LED

daniel-hofer, 2021-11-14:
Thank you for the great project.
The design of the device is very clean and I think about to use it as an cheap status light for home automation like homeassistant.
Ist it possible to control the LED colors without the particle-sensor - only to switch the colors with mqtt?
Best regards
Daniel

LordPinhead, 2021-11-14:
In theory, when we dump the transmission pin, but why, this is against the idea of the project.

daniel-hofer, 2021-11-14:
I don't want to ask for anything against the idea of the project.
I was looking for cheap and easy to use visualistaion devices. The Vindriktning is available in local stores and looks nice. Why not use it as an Reminder like - red light -> you got an email oder the laundry is finished and maybe blink-codes and so on.
I know, a simple rgb-LED on an esp8266 is an option, but so you need a case and led-diffusor-panel and so on. 
I hope i could figure out my intention.

LordPinhead, 2021-11-15:
Hm, at first thoughts, you would need to proxy the sensor and sent in your own particle data. Driving the LED itself could work too, but I have no spare one to have fun with. Maybe check the LED, and it's power with an Oscilloscope.

bvkrieken, 2021-12-20:
@daniel-hofer 
Take a look at this project:
https://www.fabiosoft.com/2021/12/02/ikea-hack-making-the-air-quality-sensor-vindriktning-smart-sensor/