Hypfer/esp8266-vindriktning-particle-sensor #41: Received message with invalid header.

CountOlaf87, 2021-11-22:
Hi,

I have uploaded to a D1 Mini, it connects to my wifi, I get the status Online message in MQTT, together with the air filter and wifi message, however, it doesn't show any values in MQTT and I can't see it in Home Assistant. When I look at the Serial output of the arduino IDE, I get this messag: "Received message with invalid header." 

Occasionally it does show a reading:

Received PM 2.5 reading: 42
Current measurements: 42, 39, 39, 40, 40

Thanks in advance

wvtbg, 2021-11-23:
This is shown when the data received from the particle meter is not as expected. Is your soldering correct and reliable?

rublinetsky, 2021-12-19:
I was having the same problem - I was getting the value from the sensor only in about 10% of tries. The rest were "Invalid header".

I've redone my wiring very carefully, tried using a voltage divider on the TXD2 pin, and tried to play with the magical delay(**15**) value (https://github.com/Hypfer/esp8266-vindriktning-particle-sensor/blob/83fd7c5034cc9737b954c5625426563492d83a2c/src/SerialCom.h#L90). Nothing helped.

Then I realized that I don't understand why we are clearing the receiving buffer in the middle of the receiving process (lines 92-94 in SerialCom.h). I commented this part out (along with the delay(15)) and it improved a lot - I am receiving reasonably looking values in more than 80% of attempts. 

Can someone explain please why we are clearing the buffer?

Thanks!

rublinetsky, 2021-12-20:
I had the same problem using ESP32 and the Hardware Serial2.

The problem is caused by the fact that, depending on how often you poll the sensor library (`SerialCom::handleUart(state)`), `serialRxBuf`receives multiple messages at a time (20 bytes each) and their combined length is often over 64 bytes. For example, I've just received `16 11 0B 00 00 00 31 00 00 03 9A 00 00 00 2F 01 00 00 00 D0 16 11 0B 00 00 00 32 00 00 03 9B 00 00 00 30 01 00 00 00 CD 16 11 0B 00 00 00 33 00 00 03 9E 00 00 00 31 01 00 00 00 C8 16 11 0B 00 00 00 33 00 00 03 9F 00 00 00 31 01 00 00 00 C7 16 11 0B 00 00 00 33 00 00 03 9F 00 00 00 31 01 00 00 00 C7 `, which is 5 messages in a row.

The code 
https://github.com/Hypfer/esp8266-vindriktning-particle-sensor/blob/83fd7c5034cc9737b954c5625426563492d83a2c/src/SerialCom.h#L92-L94
clears the buffer after 64 bytes thus deleting three whole messages (3 x 20 bytes) and the 4 bytes of the next message. What is left is not a valid message because the header was chopped off.

Anyway, the fix seems to be replacing the loop
https://github.com/Hypfer/esp8266-vindriktning-particle-sensor/blob/83fd7c5034cc9737b954c5625426563492d83a2c/src/SerialCom.h#L85-L95

with

```c++
while (sensorSerial.available() && rxBufIdx < sizeof(serialRxBuf))
{
    serialRxBuf[rxBufIdx++] = sensorSerial.read();
    Serial.print(".");
}
```

Works fine for me after this change. I've not created a pull request because I am not sure why the author used this clearing of the buffer in reading process - probably there is a good reason that I don't see.

Volcano-28, 2021-12-30:
I had the same problem. I tried to reupload sketch several times, but the problem was in incorrect configuration to the Home Assistant (mqtt). After reuploading the sensor didn't create AP again (allowing to reconfigure settings).
As I saw the mqtt configuration is stored in the flash and used even if I reupload  the sketch.
The solution is to erase the flash, like described here:
https://github.com/Hypfer/esp8266-vindriktning-particle-sensor/issues/20#issuecomment-950218909

DanielVieites93, 2023-12-04:
Hi everyone,

I'm using wemos D1 and i'm receiving this data from serialRxBuf from vindriktning:

[image]

I already try to use the sigention from @rublinetsky but still not working...
 Can you please help me?