Hypfer/esp8266-vindriktning-particle-sensor #43: Not an issue, you probably need to clean the sensor regularly

hb020, 2021-12-05:
Situation: vindriktning with small self-made esp8266 board that sends the values to my IOT system. Readout function based on this git.
Over the last 2 months the pm2.5 values have been (on average) nearly linearly increasing, up to the point where it was 140 ppm average, and red all the time. I opened up the box, found that the small filter on the sensor was all clogged up. Gave it a gentle blowout with the air compressor (yes, I protected the fan). Started all up again. The first 4-5 hours it did not seem to make any difference. But then suddenly there was a 100 ppm drop, followed by a drop of 30 more 1 hours later. Both drops were while we were out, nothing happened in the house, the device did not reboot. 

It reacts to our presence, so it is not dead, and now it is back to levels where it was when I just bought it. 
So either I damaged it, or the sensor really sensitive to dust accumulation on the sensor. I tend to believe the latter.

Maybe others also see that nearly linear increase over time? That could confirm my suspicion.

[image]

Baxxy13, 2021-12-12:
After reading this i opened my 2 sensors and removed the dust from the small intake filter. 
The measured values are now much lower, like time of first installation 3 months ago.
Thanks for the tip!
[image]

hmax42, 2022-03-08:
i have the same behavior on my 2 units
i power cycled them yesterday, which didnt happen for nearly 40 days
and the readings are back to normal

hb020, 2022-04-14:
indeed, power cycling without cleaning also significantly lowers the value. Not immediately, but within about 24 hrs. But as far as I can see, the value does not go back all the way to "when new" value.

Nismonx, 2022-04-20:
Is there a run-once self calibration instruction at boot? That would explain it.