Hypfer/esp8266-vindriktning-particle-sensor #44: 5V mention in the README

skorokithakis, 2021-12-06:
You mention in the README that the ESP8266 is 5V-tolerant, but then you mention the regulator. If you connect the 5V pad to the 5V pin of the ESP8266 board, that's not merely 5V-tolerant, it's 5V required. There's no danger of the ESP burning after a while, it'll run fine forever.

nskerl, 2021-12-17:
I was also confused by this. I have also seen other tutorials suggesting the use of a voltage divider. Perhaps there are other versions of ESP8266 board that do not have 5v pin?

dburner, 2022-01-01:
Also is this code valid for the 5V (direct link) or for the 3.3V (with voltage divider).

onshisan, 2022-01-01:
> If you connect the 5V pad to the 5V pin of the ESP8266 board, that's not merely 5V-tolerant, it's 5V required.

I understand the README to mean that the D1 is tolerant of 5V logic signals despite actually being a 3.3V logic mcu.

This project involves connecting the particle sensor to the 8266 via the IKEA PCB. The IKEA board and sensor run 5V logic, whereas [per this spec sheet the D1 is 3.3V logic for all pins/ports](https://shop.rasp.io/products/wemos-d1-mini-esp8266?variant=3485595336727). 

I gather some have added a voltage divider to protect the D1, but many have not done so and (so far) found no ill effects.

skorokithakis, 2022-01-01:
@onshisan Ahh, that makes sense, thank you. Maybe the README should add the word "logic"?