Hypfer/esp8266-vindriktning-particle-sensor #47: Not an issue, low noise mod skews sensor results by ~50%

bvkrieken, 2021-12-20:
I've implemented the low noise mod and connected the fan permanently to the 3.3v pin of the Wemos D1.
This does indeed reduce the noise significantly. But I do see a change in the sensor measurements though, they are about half as high compared to the results without the low noise mod.

[image]

My PM2,5 levels are to low to see whether this is an absolute reduction or a percentage reduction. Hopefully someone can give us a clear answer to that question.

bvkrieken, 2021-12-21:
To find out what the real difference in measurements is between a normal sensor and the sensor with the low noise mod, I've connected both setups to my domotica system and physically placed them right next to each other. Since the atmospheric circumstances are identical it should give me the exact difference in measurements.

bvkrieken, 2021-12-22:
After 24 hours of comparing the low noise modded sensor with the original sensor, I can conclude that the difference ratio between the two is ~1.82. So the 'real' value is the value of the low noise modded sensor times 1.82.
[image]

boomvalt, 2022-10-02:
This is actually a problem, because it will presumably affect the led indicator, potentially giving people a false sense of security.