Hypfer/esp8266-vindriktning-particle-sensor #52: The value does not fall below 200

tiimsvk, 2022-02-20:
Not sure how to fix incorrect display values?  He's been showing me 200-260 for some time.  And it will never go down.  I also tried to vacuum and it didn't help.

hb020, 2022-02-21:
Vacuum is not enough. I found that you need to regularly dismantle the sensor completely, and blow compressed air through the sensor (I used 3 bar, probably overkill). 
This is how far I dismantle it (just unscrew everything, no need to desolder or cut glue):
[image]

But even that no longer works lately. Trying ultrasound cleaning now.
And even then, it takes some time for the values to drop. It seems to have some memory.

hmax42, 2022-03-08:
i had a similar problem, but i solved it by power cycling the unit
https://github.com/Hypfer/esp8266-vindriktning-particle-sensor/issues/43#issuecomment-1061440029

tiimsvk, 2022-03-08:
The complete disassembly and cleaning of the sensor also helped me. Subsequently, with the passage of time, I restarted the sensor twice.
Now, after 2 weeks, the data is still fine

hmax42, 2022-03-27:
and again i powercycled my units, see the graph from one of them:
[image]