Hypfer/esp8266-vindriktning-particle-sensor #53: Web interface problem

burakcankose, 2022-03-01:
hello, first of all thanks for your development. the installation went smoothly, the sensor works fine, but the web interface is not available, when I want to change the setting, I cannot reach anywhere. I completely deleted the esp card, I reinstalled it many times, but the problem was not solved, what can I do?

fusebyte, 2022-08-19:
I cannot enter web interface too, even if i try to access it manually

AseKarlsson, 2023-02-12:
Hi, Did you solve it, I am in the same situation?

tabearu, 2023-02-12:
Were you at some point able to connect the device to your WiFi? I figured out that as soon as the WiFi connects, there's no way to change any settings, even if for example MQTT-settings are wrong. I had this issue and you can 'solve' it by removing the connection to your WiFi. This way, the esp will go back to AP-mode. Easiest solution is to turn off your WiFi until you've changed the settings. If that's not possible try wrapping a bit of aluminium foil around the esp to block the WiFi connection. Can be a bit tricky to figure out how much foil you need in order to be able to connect (e.g. with a phone) and block the WiFi but it worked for me.

AseKarlsson, 2023-02-13:
I am not able to connect to web interface at all, not a single time.
Probably due to Issue #61