Hypfer/esp8266-vindriktning-particle-sensor #55: Embarrasingly stuck

Nukle0n, 2022-07-17:
So I set up an MQTT broker and linked it to Home Assistant, and I soldered the board in and flashed it, and ran the config from its ad-hoc network... now what? I'm seeing the device on my router but I don't know what to do next, none of the guides on MQTT broker stuff seem to explain what to do, if I should just expect it to work? I tried inputting the address of my MQTT broke both with and without port number and I don't really know what the heck to do. Do i need to broadcast a topic? What even is a topic? Should I have set something in the code before flashing it?

My apologies for ranting but this is has turned out to be a really poor example of a first time MQTT project 😅

Rclark01, 2022-08-24:
I’m stuck here too 😖

Were you able to get past this?

Nukle0n, 2022-08-24:
You need to make sure your MQTT broker is accessible, you have to set up the "listener" argument in the .conf file so it can be accessed outside of the local machine

Rclark01, 2022-08-24:
Thank you very much! I got it working this afternoon

boomvalt, 2022-10-02:
Sorry for the newbie question, but where do I find the .conf file?

Nukle0n, 2022-10-02:
> Sorry for the newbie question, but where do I find the .conf file?

If you're using HAOS there's no conf file, you just need to set up MQTT as an extension. If you're doing docker you need map a static directory to where there is a conf file.