Hypfer/esp8266-vindriktning-particle-sensor #61: Version Questions on Library Manager

AseKarlsson, 2023-02-07:
Hi,

For Library Manager it says versions:

ArduinoOTA 1.0.3
ArduinoJSON 6.10.1
PubSubClient 2.8.0
WiFiManager 0.15.0

Does it have to be these or do newer work as well?

Thanks,
AseKarlsson

AseKarlsson, 2023-02-10:
Hi,
Can there also be info hos Libraries?
If I serach on theese I get a dosen?
And I fail to install

//AseKarlsson

AseKarlsson, 2023-02-10:
Tried with these exact versions but it fails, any help?

AseKarlsson, 2023-02-11:
Installed:
	ArduinoJSON by Benoit Blanchon, version 6.20.1
	WifiManager by tzapu, version 0.15.0
	PubSubClient by Nick OLeary, version 2.8.0

Opened the .ino file and let it create folder and copy itself into it.
Copied the following files into the newly sreated folder:
	Config.h
	SerialCom.h
	Types.h

Loaded the project and when verify:
c:\Downloads\Arduino\libraries\WhareHauoraWiFiManager/WiFiManager.h:25:24: error: 'const char HTTP_HEAD []' redeclared as different kind of entity
   25 | const char HTTP_HEAD[] PROGMEM            = "<!DOCTYPE html><html lang=\"en\"><head><meta name=\"viewport\" content=\"width=device-width, initial-scale=1, user-scalable=no\"/><title>{v}</title>";
      |                        ^~~~~~~
In file included from C:\Downloads\Arduino\esp8266-vindriktning-particle-sensor\esp8266-vindriktning-particle-sensor.ino:4:
C:\Users\XXX\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1\libraries\ESP8266WebServer\src/ESP8266WebServer.h:47:39: note: previous declaration 'HTTPMethod HTTP_HEAD'
   47 | enum HTTPMethod { HTTP_ANY, HTTP_GET, HTTP_HEAD, HTTP_POST, HTTP_PUT, HTTP_PATCH, HTTP_DELETE, HTTP_OPTIONS };
      |                                       ^~~~~~~~~

exit status 1

Compilation error: exit status 1

Any help appreciated!
Thanks,
AseKarlsson

AseKarlsson, 2023-02-11:
Some more info after googeling other projects. If I change in WifiManagers WiFiManager.h file the line HTTP_HEAD to HTTP_HEADS it and compile. I can see the unit now on the network, but cannot connect.

AseKarlsson, 2023-02-12:
Hi,

Is this project and forum all dead?

I failed to compile the original code.
Found the HTTP_HEAD to HTTP_HEADS from other projects but they fail as well to connect to the unit as me.

Any help, would prefer MQTT before ESP Home

Thanks,
AseKarlsson

AseKarlsson, 2023-02-14:
@Hypfer, Can you please take a look if you can understand why I cannot compile?

AseKarlsson, 2023-02-14:
> Do you have a support contract?

Nope, just hoping for help :)

Hypfer, 2023-02-14:
*demanding