Hypfer/esp8266-vindriktning-particle-sensor #62: How to See the Data in an App ?

Codebysolder, 2023-03-06:
I have the Code running OK on The ESP8266 D1 Mini in the VINDRIKTNING.
The Serial Port is streaming measurements to my IDE Serial Monitor.

What do I have to  use to see the data on an App ? (or a browser)
The Config.h is untouched, I dont know what to enter !
It has somehow managed to connect to my WiFi despite not giving it an SSID or Password.

I installed Home Assistant on my Android Tablet, it asks for a URL, I dont have one
The Blog does say this code  works with auto discovery, but no joy.

This must be a pretty basic issue, but I cant find any tutorial related to it.

Codebysolder, 2023-03-07:
Hmm, no takers.
Its a shame that for the sake of a basic tutorial on setting this up that such a complex program is likely to be abandoned.
Looking at the other Issues, I am not the only person at a loss to getting this finished.

I have done numerous other ESP projects using Blynk and Thingspeak, so I will have a go at using the  data in the serial feed to  feed Thingspeak - has anyone else done this ?

I want to use a simple manual configuration of ssid/password for the wifi, so first task is to clear out all the code around this.

I added a Piezo buzzer between port D6 and +3.3V, very useful to indicate the WiFi has connected or whatever you want.
I attach a few pics of a simple way of wiring this up using normal 1/10th  inch pitch headers, tweaked so they surface mount and then supported with epoxy resin (not hot melt). I used double sided tape to fix the board (with a 3mm spacer) with the socket available for external  connection, easy to reflash with it all screwed together.

[image]
[image]
[image]

[image]
[image]
[image]
[image]
[image]

Codebysolder, 2023-03-07:
After  remming out about half the code and adding in Thingspeak, its working.
[image]

Nismonx, 2023-03-07:
Probably worth doing some reading on MQTT....
Homeassistant, nodered....

Codebysolder, 2023-03-07:
I accept that, I did look at the mqtt site and the "quick start" seemed anything but !
I have used Thingspeak for a few projects and it works very well, I prefer it to Blynk.
The code in this project is well beyond my capability, I am not a natural programmer.