Hypfer/esp8266-vindriktning-particle-sensor #64: Also working with the new version, VINDSTYRKA

mStirner, 2024-07-06:
Does some one have tried this project with the new version VINDSTYRKA?
https://www.ikea.com/de/de/p/vindstyrka-luftqualitaetssensor-smart-00498231/

LordPinhead, 2024-07-06:
I know from this project it was hacked at least https://github.com/oleksiikutuzov/IKEA-VINDSTYRKA/issues/1 and it has a standard sensor, at least that is what I've read. I have no Vindstyrka, otherwise, I would've tried it with Tasmota or ESPHome already.

LordPinhead, 2024-07-21:
So, FYI, I installed an ESP32 S2 board into the Sensor.
Pins: 5V GND, GPIO21 for SDA and GPIO22 SCL
Working flawlessly with Home Assistant or even standalone when you know how to Configure MQTT.
Config (api keys changed):

```
esphome:
  name: vindstyrka
  friendly_name: Vindstyrka

esp32:
  board: nodemcu-32s
  framework:
    type: arduino

logger:

api:
  encryption:
    key: "8cD8IXw54kcq43qgaH6kFaJWid3WjciK3mHVKkwdSO8="

ota:
  - platform: esphome
    password: "7d3b62f63e1417f1f07f7195ad4cc8c4"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  domain: .your-internal-domain

  ap:
    ssid: "Supersensor Fallback Hotspot"
    password: "fallbackpw"

i2c:
  sda: 21
  scl: 22
  scan: true
  id: bus_a

sensor:
  - platform: sen5x
    id: sen54
    i2c_id: bus_a
    pm_1_0:
      name: " PM <1µm Weight concentration"
      id: pm_1_0
      accuracy_decimals: 1
    pm_2_5:
      name: " PM <2.5µm Weight concentration"
      id: pm_2_5
      accuracy_decimals: 1
    pm_4_0:
      name: " PM <4µm Weight concentration"
      id: pm_4_0
      accuracy_decimals: 1
    pm_10_0:
      name: " PM <10µm Weight concentration"
      id: pm_10_0
      accuracy_decimals: 1
    temperature:
      name: "Temperature"
      accuracy_decimals: 1
    humidity:
      name: "Humidity"
      accuracy_decimals: 0
    voc:
      name: "VOC"
      algorithm_tuning:
        index_offset: 100
        learning_time_offset_hours: 12
        learning_time_gain_hours: 12
        gating_max_duration_minutes: 180
        std_initial: 50
        gain_factor: 230
    temperature_compensation:
      offset: 0
      normalized_offset_slope: 0
      time_constant: 0
    acceleration_mode: low
    store_baseline: true
    address: 0x69
    update_interval: 10s
```

Don't solder a pinheader on, it's not enough space in the sensor case for it, solder the wires directly onto the ESP.